RobTillaart/MCP_ADC #3: This does not seem to work when using the ESP32

intakenick, 2021-02-19:
First off, thanks for sharing this! I'm having a few issues:

When reading from channel 0, the results are exactly 1/2 of what I'm measuring on my scope. This does not change if I alter the SPI frequency

When reading from channel 1, it always returns zero.

Using other libs for this chip, I always get skewed values, this has been most consistent despite still not providing the expected values. Any help is greatly appreciated.

RobTillaart, 2021-02-19:
Will check tomorrow, too busy today.
Which mcp do you use?
Possibly i can recreate your setup.

intakenick, 2021-02-19:
I'm using the MCP3002

coyan, 2021-02-19:
There is any reason to force to use the default SPI port? Some boards like Arduino Mega or the ESP32 have more than one SPI port.

I found this file and can be changed easy to work with the VSPI:
https://github.com/MIDI-CHIP/MCP3008/blob/master/src/mcp3008.cpp

But I think that can be better to use the SPI as an object, for example on this topic they talk about it:
https://forum.arduino.cc/index.php?topic=688279.0

Maybe a easy way to implement is using something like this example and a single preprocessor define (by default uses the SPI like now and when we define something will use the vspi/hspi alternative):
https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/examples/SPI_Multiple_Buses/SPI_Multiple_Buses.ino

If you like the idea I can help to debug with a ESP32 and the MCP3008.

RobTillaart, 2021-02-19:
The reason for the default SPI port? It is a starting point. First get something working and add functionality in small steps.
I need to build up more ESP experience so your links and your offer for help are definitely appreciated.

RobTillaart, 2021-02-20:
(my agenda filled up with priority things so expect delay's)

### Problem 1: measurement too low
You state you get only half the value expected with channel 0.

**Can you tell more about your measurement setup?** 
- ESP32 + MCP3002
- voltage of the MCP3002 =  3V3  or 5V ?
- voltage to be measured?
- Are the grounds all connected for same reference?

**Can you make 3 measurements with channel 0 ?**
  - print the HEX values of the bytes received in **readADC( )**   (around line 95- 100 in MCP_ADC.cpp) 
  -  when applying  0V, 3V3, something in between.
  
The MCP3002 is a 10 bit so some of the expected values are  -  there might be noise
| voltage | hibyte | lobyte |
|:----:|:----:|:----:|
|  0V  |   0x00  |  0x00  |
| 1V  |   0x01 | 0x36 |
|  2V  | 0x02 |  0x6C |
|  3V3 |  0x03 | 0xFF |

When using 5V to power the MCP3002 (= reference voltage) these numbers will be different.

RobTillaart, 2021-02-24:
(found some time)
Did a test with an MCP3008 and an UNO (no ESP32 nearby) and connected it to 5V - both VDD and Vref
This mimics the MCP3002 which does not have separate VDD/Vref

5V ==> 1021-1023 as expected  => note more noise than expected.
GND ==>  0 as expected
3V3 ==> 666-667 as expected

Powering the MCP3008 with 3V3 gave the following results:
GND ==> 0 - expected
3V3  ==> 1021 - very stable but expected 1023 
5V  ==>  1023 - very stable (don't know if this is seen as an over voltage but at least the value is as expected.

So although not tested with ESP32, the library acts like it should.

Please note that the actual reading is affected (~33%) by the value of the (VDD/) Vref pin.

BootsSiR, 2021-03-10:
@intakenick any luck getting good readings from this?  I have an MCP3208 connected to an ESP32 and my values are low as well.  I have a simple photoresistor hooked up to channel 0 and I see a range from 255-1250 on the MCP3208 with this lib and 850-4096 on the ESP32 ADC.  I haven't had any luck with any other libraries working at all.

intakenick, 2021-03-10:
I had some luck using the MCP3XXX library by Chris Baker. I had to lower
the SPI frequency within the h file to get more accurate results.

Line 278: typedef MCP3XXX_<10, 2, 1200000, 2> MCP3002;

I changed 1200000 (1.2MHz) to 100000 (100kHz) which is plenty fast enough
for my application. Results get more accurate with longer sampling time,
I've gone down as low as 10000.

Hope this helps.

-N

--

Nick Johnson
Intake

(914) 906-3527 [mobile]

***@***.***
------------------------------------------------------------------------------------------------

Automating preventive care. (myintakepro.com)

NOTICE OF TRANSMISSION: This email and any files transmitted with it are
confidential and intended solely for the use of the individual or entity to
whom they are addressed. If you are not the named addressee please notify
the sender immediately and delete this e-mail from your system. If you are
not the intended recipient you are notified that disclosing, copying,
distributing or taking any action in reliance on the contents of this
information is strictly prohibited.

RobTillaart, 2021-03-11:
@intakenick 
interesting find, it would mean that software SPI, which is slower, should result in more accurate measurements.

I definitely need find some time to verify / test.

BootsSiR, 2021-03-11:
I ended up using a different library to prototype with but I'm going to revisit this one when my final boards are complete as it seems to be the most actively maintained  👍

RobTillaart, 2021-03-12:
Do not have a working setup yet, but found a probable cause for the MCP3002

The buildRequest is not correct (line 132 in .cpp file)   
Code below fixes OR when single 

can you please check if this improves the measurement?
```cpp
uint8_t MCP3002::buildRequest(uint8_t channel, bool single, uint8_t * data)
{
    // P17  fig 6.1   MCP3002 
  data[0] = 0x44;                          // start bit + MSB first bit
  if (single) data[0] |= 0x20;             // single read | differential
  if (channel) data[0] |= (channel << 4);  // channel = 0 or 1;
  return 2;
}
```

RobTillaart, 2021-03-12:
Got my MCP3008 working with software SPI  

- DOIT ESP32 DEVKIT V1
- MPC3008
- software SPI
```
MCP3008 mcp1(23, 19, 21);  // dataIn, datOut, Clock
```

Read takes < 15ms
All 8 channels OK (tested with a potmeter)
What I noticed on the scope was that if the last bit is 1 it is not pulled to GND by the MCP3008.
It follows a nice "unloading" RC curve

BootsSiR, 2021-03-12:
@RobTillaart I have the MCP3208 (12 bit version) .. will your changes affect that?

RobTillaart, 2021-03-12:
Screenshot signal software SPI

[image]

Very visible is the floating line at the end.
Also good visible are quite nice square pulses

RobTillaart, 2021-03-12:
Screenshot signal hardware SPI  (VSPI pins)
[image]

Idem floating at the end,
The pulses are not clean, not the clock and not the data - just crap imho.
I will try to lower the speed of the HW SPI to see if that helps

RobTillaart, 2021-03-12:
> @RobTillaart I have the MCP3208 (12 bit version) .. will your changes affect that?

The changes posted above with the OR fix is only for the MCP3002. 
I need to review the **buildRequest()** function of the MCP3208 and check the datasheet.
I'll be back

BootsSiR, 2021-03-12:
@RobTillaart Appreciate the work!  I'll test the 3208 when ready.

RobTillaart, 2021-03-12:
@BootsSiR 
@intakenick

A review showed that the bit pattern created in the  **buildRequest()** for the MCP3208 is good.
Problem could be the signal deformation of the HW SPI (which might be a breadboard issue but still)

Can you verify that the software SPI modus is working correctly? 

----
Created a branch named: ESP32demo  with the fix for the mcp3002 and 2 demo sketches specific for the ESP32.

The ESP32 software SPI is working correctly with the MPC3008 , 
The ESP32 hardware SPI is under investigation.

BootsSiR, 2021-03-12:
@RobTillaart 

For reference, I seem to get good values using hardware SPI and this library:

https://github.com/labfruits/mcp320x

RobTillaart, 2021-03-12:
Thanks, I'll have a look if they reduce the speed somehow (which is my assumption)

RobTillaart, 2021-03-12:
Yep, labfruits use 1.6 MHz, I use 16MHz,

<some testing>

8 MHz fails, and it misses bits resulting in lower values.
6MHz seems to work,  
4MHz is safe, ==> will make that the default value in the library

I will update my ESP32 hardware SPI example with a extra line to **setSPISpeed()**

RobTillaart, 2021-03-12:
The datasheet MCP3208 page 3  states for 2.7V (~~ESP32) a max speed of 1 MHz, 
so even the software SPI on ESP32 might be too fast...

The 4 MHz is about the 250 ns mentioned. 

So after 6 hours of analysis it all makes sense :) 
Now fix the library and the readme.

[image]

BootsSiR, 2021-03-12:
Good find 👍

RobTillaart, 2021-03-12:
Thanks

Signal at 4 MHz looks quite good.

[image]

RobTillaart, 2021-03-12:
Updated ESP32demo branch with working HW SPI @ 4 MHz

BootsSiR, 2021-03-12:
Just merged it?  Branch disappeared before I could pull it.  Going to test now :)

RobTillaart, 2021-03-12:
ESP32demo branch is merged and version 0.1.2 is released.

BootsSiR, 2021-03-12:
@RobTillaart the values I'm reading now line up with the values I was getting using the labfruits library.  Thanks for the fixes!

RobTillaart, 2021-03-12:
Good to hear!
The read performance will be a lower, but quality is more important than speed 😀

RobTillaart, 2021-03-12:
If you are missing things in this or one of my other libraries (or missing a library at all) just let me know

RobTillaart, 2021-03-12:
@intakenick

Can you confirm it works now for you too?

RobTillaart, 2021-03-20:
If problem reappears please reopen issue.

closed 2021-03-20 by RobTillaart