RobTillaart/MCP_ADC #5: ESP32 hspi & vspi

RodEnry, 2021-05-20:
Hi Rob, 
thank you for your awesome library!
Do you think it's possible to address the library to a specif SPI bus?

I'm writing to an SD using the classic SPI library. I initialize the SD card like:
`SPI.begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS_PIN);`

About the ADC, I'm using this initialization, which is working: 
   ```
 hspi = new SPIClass(HSPI);
 pinMode(ADC_CS_PIN, OUTPUT);
 digitalWrite(ADC_CS_PIN, HIGH);
 digitalWrite(ADC_CS_PIN, LOW);
 delay(10);
 hspi->begin(ADC_SCK, ADC_MISO, ADC_MOSI, ADC_CS_PIN);
 hspi->setFrequency(ADC_CLK);        //1600000
 hspi->setDataMode(SPI_MODE0);       //SPIMODE 0
 hspi->setBitOrder(MSBFIRST);   
```  

BUT, I'm obtaining a lot of spikes in the ACD readings, let's say at 500Hz.
So, since the SPI is already used, do you have any idea how to integrate your library in this way?

Cheers
Enry

RobTillaart, 2021-05-20:
The idea of SPI is that you share CLK MISO and MOSI lines. So only the CS line should be different. 

That works well until different tasks want to access both devices ADC and SD in a single moment. One solution is to have a semaphore on the SPI bus.
Another is to add a message queue for the SD card so it can write between sampling.
Third solution is to use two SPI busses side by side. Optional a SWSPI for one and HWSPI for the other. Which what depends on the amount of data to move.

RobTillaart, 2021-05-20:
Spikes
Check wiring and optionally add a low pass filter before the ADC (resistor, capacitor). Or take median of e.g. 7 samples to reduce spikes.

RodEnry, 2021-05-20:
Hi Rob, 
thanks for your prompt reply!

I already added a low pass filter. The spikes are not caused by some external noise, but they're coming from the SPI comunication itself.

I'm currently using two SPI buses at the same time (including a semaphore on the SD card), mainly because the tasks are very demanding.

RobTillaart, 2021-05-20:
Do you have a scope to check the signal?
It also might be that you need to adjust pull up resistors on the CLK an data lines.

Is it really needed to set SPI bus to 16 MHz. No datasheet nearby but I recall 2 or 4 Mbit is max supported. Higher speeds could cause faulty bits...

RobTillaart, 2021-06-07:
Any progress?

RodEnry, 2021-06-07:
Hi Rob, 
unfortunately I didn't have a scope to check the signal. 

But I found a workaround!
Basically, I'm doing several reads (x3) every millisecond.
After tweaking the adc functions I was able to read the data correctly.
In particular the spiky one was the first reading! I added a "fake" first reading and now everything is clean. Maybe an issue on the sample & hold capacitor inside the chip?

RobTillaart, 2021-06-07:
> unfortunately I didn't have a scope to check the signal.

look for the - JYE Tech DSO150 - not a super profi scope, very affordable and easy to take with you. 
Worth the money imho.

> Maybe an issue on the sample & hold capacitor inside the chip?

Could be, but why only the first of three?  It is strange. 

> But I found a workaround!

Can we consider the issue closed then?

RodEnry, 2021-06-08:
Yes! Many thanks!

Enry

closed 2021-06-08 by RodEnry