RobTillaart/MCP_ADC #13: 3201 tested

johedlund, 2023-09-07:
Hi
I have tested 2.0 on the 3201 and ESP32 using HWSPI but I keep getting rollover values when ADC-IN is at half of Vref (3V3 in my case)
So the values goes from 0-4095-0-4095 when doing 0-3V3 on ADC-IN.

I tried the direct SPI reading of the ADC and got it working with these lines:

digitalWrite(CS, LOW);
    

    // Read MSB and LSB
    uint16_t msb = SPI2.transfer(0x00);
    uint16_t lsb = SPI2.transfer(0x00);

    // Extract relevant bits: Keep 5 MSB bits and 7 LSB bits
    uint16_t masked_msb = (msb & 0x1F);
    uint16_t masked_lsb = (lsb & 0x7F);

    // Combine and align bits to form 12-bit value
    ADC_VALUE = (masked_msb << 7) | masked_lsb;

    digitalWrite(CS, HIGH);

RobTillaart, 2023-09-07:
Thanks for this feedback of the MCP3201, it is the one I did not test yet.

Tomorrow I will investigate where/how the library differs and fails. 
Given your description I suspect some bit magic error, 

Again thanks for reporting!

RobTillaart, 2023-09-07:
Could you do a test with the library for me?
(When time permits)
Use the ESP32 and the SW SPI option?
Does it show the same problem?
Thanks

johedlund, 2023-09-07:
I would like to help but currently my only components are soldered to a pcb. So it is difficult to try other connecting pins

RobTillaart, 2023-09-07:
No problem, 
I will just do thought experiments then, can solve a lot.
(Einstein was successful with it with far more difficult matter :)

to be continued !

RobTillaart, 2023-09-08:
What is possible to test is using the HSPI pins as SWSPI pins.

johedlund, 2023-09-08:
Ah, of course. I might be able to test that when time permits. Thanks for fast feedback.

RobTillaart, 2023-09-08:
Q: are there other SPI devices on the bus?
If so do they use a certain SPI setting?

My first investigation of the datasheet shows this picture

[image]

This clearly shows that the 12 data bits are send one after another. 
Note: your working code you effectively skips 1 bit (by masking 0x7F) so that is not in line with datasheet.
I come to the following hypothesis. 

#### Hypothesis

If the data bit are read 1 clock too early, the library effectively multiplies the raw value read by 2.
This would give a raw value range from 0 to 8191 
However as the library masks the raw value with 0x0FFF / 4095 / maxValue
```cpp
if (bytes == 2) return ((256 * data[0] + data[1]) & _maxValue);
```
the raw range of 0..8191 will appear as 0..4095 - 0..4095 as the highest bit is masked away.

This hypothesis would explain your observation with the library.

Reading the data bit too early can be caused by different mode.

investigation continues later today.

johedlund, 2023-09-08:
Well, digital communicaitonis not my speciality as I only tinker with small hobby projects. I needed to get the ADC working in my setup and tried other ways than the library to get out 0-4095.
There are no other SPI devices in the circuit. I have tried different SPI speeds from 400 kHz to 4 MHz but see no difference there.

RobTillaart, 2023-09-08:
Will think of a library patch today.
If you divide the raw value by 2 before the raw value is masked you should get 0..4095.
Not a definitive solution but might work.
Later this afternoon

RobTillaart, 2023-09-08:
Read the datasheet a bit further and found this picture 
It shows the bits are clocked in one clock pulse sooner than e.g. the MCP3202

### MCP3201
[image]

Note the extra B1 at the end.
Same is true for the MCP3001 (not supported yet but will be soon).

### MCP3202
[image]

RobTillaart, 2023-09-08:
@johedlund 
Please note that **readADCMultiple()** is not patched yet.
Working on the solution, expect in an hour or so.

RobTillaart, 2023-09-08:
Develop branch created, build is running.

RobTillaart, 2023-09-08:
@johedlund 
Can you please test the development branch if possible?

johedlund, 2023-09-09:
Yes will do but am out sailing this weekend.

RobTillaart, 2023-09-09:
Have a good wind!

johedlund, 2023-09-11:
Tested today with the dev branch 0.2.1 and the bug with rollover values is gone. I get a full 0..4095 on a full turn of a potentiometer connected to ADC_IN. 
I retested with the current Master 0.2.0 and I am still able to reproduce the bug with that version. 
Good work!

RobTillaart, 2023-09-11:
Super, thanks for testing
Will merge asap.

closed 2023-09-11 by RobTillaart

RobTillaart, 2023-09-11:
@johedlund 
Again thanks for testing the 3201, code is merged into master and should be available in library manager next 24 hrs.
(Or download directly from GitHub or PlatformIO)