RobTillaart/MCP_ADC #20: Library for the faster ADC081Sxxx ADC chips

avandalen, 2024-01-09:
Dear Rob, 

Do you think the library also works for the ADC081S chips? These chips also work with SPI but are faster and have also SOT23-6 packages.

Best regards,
Albert

https://avdweb.nl/arduino

RobTillaart, 2024-01-10:
Thanks for the issue,
Have to dive into the datasheet to verify.  The device might be part of a family so the question needs possibly a broader answer.
Will look into it later this week.

RobTillaart, 2024-01-10:
Found these datasheets
- https://www.ti.com/lit/ds/symlink/adc081c021.pdf  <<< has far more information.
- https://www.ti.com/lit/ds/symlink/adc081s021.pdf

There are different related chips with resolutions 8,10, 12 bit, extra address pins and/or alert function.

A quick compare shows that the ADC081S series has registers to set where the mcp_adc has not.

So they are not compatible

RobTillaart, 2024-01-10:
@avandalen 
As far as I know there is no Arduino library for this ADC081S device 
(I made a note on my backlog, between ~50 other missing ones).

RobTillaart, 2024-01-11:
mmm, a 2nd look, ==> the C-021 == I2C  and S-021 == SPI.

Still not compatible :(  but the SPI version seems pretty straightforward.