RobTillaart/MCP_ADC #21: Unclear instructions on how to use this with ESP32 S2

khangtranduc, 2024-01-18:
Set up code
```cpp
#define SPI_MOSI 11
#define SPI_MISO 13
#define SPI_CLK 12
#define SPI_CS1 10

MCP3208 adc1(SPI_MOSI, SPI_MISO, SPI_CLK);

void setup(){
     adc1.setSPIspeed(4e6);
     adc1.begin(10);
}
```

adc1.read(0) does not return the expected value. Previous version of the library (0.2.1) worked fine when I used it in the past, now no longer works. New library does not work.

Unclear instructions on how to use both HWSPI and SWSPI with this library. ESP32 example is outdated.

RobTillaart, 2024-01-18:
Thanks for this issue.
busy fixing (prio) bugs so it will take some time before I can dive into this one.

RobTillaart, 2024-01-18:
> adc1.read(0) does not return the expected value. 
Previous version of the library (0.2.1) worked fine when I used it in the past, now no longer works. 
New library does not work.

Both the version 0.3.0 and 0.4.0 have introduced breaking changes. 
These are documented in the readme.md file.

Both version did not change the fetching of data from the ADC.

You are using the SW SPI constructor and I do not know if the GPIO 10-13 pins are free to use. 
They look like the HW SPI pins of an UNO.

Can you try these pins? They are from the hardware VSPI, but can be used for SW SPI too.
//  For VSPI (id = 2):
//    CLK:  18,
//    MOSI: 23,
//    MISO: 19,

khangtranduc, 2024-01-18:
Thank you for your prompt response, I really appreciate it!
The pins I am using are from the [ESP32 S2 Saola 1](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-saola-1-v1.2.html) board, there are two sets of HWSPI pins one from 10-13 and another from 34-37. 

I was able to use pins 10-13 to control a MCP3201 using version 0.2.1. of the library. I will take another look at the changes documented tomorrow. Will update if there's any new developments!

RobTillaart, 2024-01-18:
You should be able to use one of the hardware SPI by using it implicitly
```cpp
MCP3208 adc1;

void setup(){
     adc1.setSPIspeed(4e6);
     adc1.begin(10);
}
```

or I assume using FSPI as I do not see HSPI/VSPI pins mentioned

```cpp
MCP3208 adc1(&FSPI);

void setup(){
     adc1.setSPIspeed(4e6);
     adc1.begin(10);
}
```

(from your link)

[image]

khangtranduc, 2024-01-19:
Hi Rob,
I took another look at the project today, I managed this get the new library to work for MCP3201; however, I am running into trouble trying to get a similar code to work for MCP3208. Do you have any idea why that might be the case?
```cpp
#include "Arduino.h"

#include "MCP_ADC.h"

#define MOSI 11
#define MISO 13
#define CLK 12
#define CS 10

SPIClass * hspi = new SPIClass(HSPI);
MCP3201 adc(hspi);

void setup() {
    Serial.begin(115200);

    hspi->begin(CLK, MISO, MOSI, CS);
    
    adc.begin(CS);
}

void loop() {
    Serial.println(5.0 * ((float) adc.read(0)/adc.maxValue()));
}
```

khangtranduc, 2024-01-19:
Oh I realized my mistake, the MCP3208 was on an SPI Bus and there were some interference. It works now
On another note, in the begin function, I think its best to not call _mySPI->begin() for hw mode. As there are devices such as the ESP32 which can do hw spi for any GPIO pins and would require a special begin call with all the SPI pins specified as shown in my previous response.
```cpp
void MCP_ADC::begin(uint8_t select)
{
  _select = select;
  pinMode(_select, OUTPUT);
  digitalWrite(_select, HIGH);
  digitalWrite(_select, LOW);    //  force communication See datasheet)
  digitalWrite(_select, HIGH);

  _spi_settings = SPISettings(_SPIspeed, MSBFIRST, SPI_MODE0);

  if (_hwSPI)
  {
    // _mySPI->end(); - These should be removed, user to begin their own SPI like the Wire in your PCA9635 library.
    // _mySPI->begin();
  }
  else                 //  software SPI
  {
    pinMode(_dataIn,  INPUT);
    pinMode(_dataOut, OUTPUT);
    pinMode(_clock,   OUTPUT);
    digitalWrite(_dataOut, LOW);
    digitalWrite(_clock,   LOW);
  }
}
```

RobTillaart, 2024-01-19:
@khangtranduc 
>On another note, in the begin function, I think its best to not call _mySPI->begin() for hw mode. As there are devices such as the ESP32 which can do hw spi for any GPIO pins and would require a special begin call with all the SPI pins specified as shown in my previous response.

Think you have a point here as the "internal" **begin()** call would possibly use default parameters. 
So that implies to ~20 libraries, so some work to be done.

RobTillaart, 2024-01-19:
Created a separate issue for it #22

RobTillaart, 2024-01-19:
@khangtranduc 
As you got the ESP32 running I assume this issue can be closed.

closed 2024-01-22 by khangtranduc